toolboc/psx-pi-smbshare #2: exfat compability?

Clydefrosch22, 2018-08-10:
In order to avoid having to split backups over 4gb in fat32 and prevent too many unneeded writes in ntfs, I reformated my thumbdrive to exfat, resulting in an error in OPL.

Is exfat simply not supported, or should I reformat again/use a special formatter to make it work?

Drive is a sandisk ultra flair with 128gb

toolboc, 2018-08-12:
Thanks for confirming that this is an exfat support issue.  

Would you be able to try the suggestions mentioned in this [thread](https://www.raspberrypi.org/forums/viewtopic.php?t=180798)?  

My guess is that it * might * work with:
`apt-get install exfat-fuse exfat-utils`

If it does, I'm happy to accept a PR or make the change myself if you can provide confirmation.

Thank you!

Clydefrosch22, 2018-08-12:
I will, but I'm not exactly sure how.
I connected the pi to my tv and when its done loading, it shows some kind of log in promt P think.
Do i just type that line once that happens? 

I'll try finding a working keyboard as soon as I can.

Reading the thread, they also mention read only ntfs support, so that's probably responsible for the other issue I posted, might even be fixed with this.

toolboc, 2018-08-12:
@Clydefrosch22 ,

No worries, happy to help!

You will need to ssh to the pi to run the command.  To do this, yoy will just need to obtain the ip of the device and follow the instructions in step 4 [here](https://www.raspberrypi.org/documentation/remote-access/ssh/).  Steps 1-3 have already been done in the presupplied image so they are safe to skip.

The pi is configured with standard login defaults, i.e. user: `pi` / password: `raspberry`

I apologize in putting the work on you for testing this potential solution, but it's really the best way to independently verify the solution given your setup.  

I agree that there is a really good chance that this might also resolve the other issue.

Let us know what happens and feel free to reach out if you need additional help!

Clydefrosch22, 2018-08-12:
ok, found a keyboard, understood that it was set to us layout and finally got the command in, but its giving me an error
https://imgur.com/a/R9wMsJt

toolboc, 2018-08-12:
Ah, just add `sudo` to the command i.e. `sudo apt-get install exfat-fuse exfat-utils`
then type the password `raspberry` if prompted and hit enter.

the8thsign, 2018-08-13:
I gave this a try, ExFAT USB is not mounting after installing exfat-fuse and exfat-utils

toolboc, 2018-08-13:
I'm noticing alot of mixed results online about ExFAT compatibility on Raspbian.

The best advice for now is to use FAT32 if possible. 

I can try to look deeper later in the week but am not feeling very good at the moment since exfat-fuse and exfat-utils aren't working as-is.

If anyone has any additional suggestions, it would definitely help.

The relevant code that supports USB drive auto-mounting can be found [here](https://github.com/toolboc/psx-pi-smbshare/blob/master/automount-usb.sh) if anyone wants to take a look.

Clydefrosch22, 2018-08-13:
Unfortunately, that's not working for me.
first command got some error about not finding an exfat package, then ./configure and everything past that got errors too

toolboc, 2018-08-13:
@Clydefrosch22 ,

It looks like we need a few more dependencies and a bit of modification to the steps I posted in my previous reply.

I arrive home later today and should be able to try things out on my device.  If anyone wants to get a head start, take a look at [autogen.sh](https://github.com/stigi/pmount-exfat/blob/master/autogen.sh) in sigi's repo.  

Looks like we need to run that script and have auotmake, aclocal, and intltoolize installed to successfully build pmount from source.

I'll update later today with working instructions and then we can try to see if it works on your device.

toolboc, 2018-08-14:
ExFat and NTFS support is now working!  

To update your existing image, simply run the following:
   
    # Update packages
    sudo apt-get update

    # Add NTFS Read/Write Support
    sudo apt-get install -y ntfs-3g 

    # Add ExFAT support
    sudo apt-get install -y exfat-fuse exfat-utils autoconf intltool libtool libtool-bin libglib2.0-dev libblkid=dev
    cd ~
    git clone https://github.com/stigi/pmount-exfat.git
    cd pmount-exfat
    ./autogen.sh
    make
    sudo make install prefix=usr
    sudo sed -i 's/not_physically_logged_allow = no/not_physically_logged_allow = yes/' /etc/pmount.conf

These changes will be added to a new release later tonight, I will update this post when it is ready.

closed 2018-08-14 by toolboc