toolboc/psx-pi-smbshare #3: Error trying to save game configs (like Modes)

Clydefrosch22, 2018-08-11:
Ok, final issue, I swear.
Trying to save game configurations (triangle menu in OPL) produces an error "Error writing settings!", potentially because there's only reading permissions set?

toolboc, 2018-08-14:
@Clydefrosch22 , can you double-check that this issue is also resolved by the fix supplied for #2 ?

toolboc, 2018-08-14:
@Clydefrosch22 ,

Awesome!  Thanks for being an awesome user and for assisting in testing and development!  Always happy to help!

closed 2018-08-14 by toolboc

Clydefrosch22, 2018-08-18:
I'm not sure if I'll be able to contribute much else, the only other thing I had a few minutes of mild issues with was setting up wifi cause I kept typing ssid and password in "<>"  without removing <>.

Everything else, at least as far as using this for a ps2, was as self-explanatory as it would get for someone new to both ps2 hacking and using a raspberry pi.