toolboc/psx-pi-smbshare #4: USB Mounts are not setup with folder structure

the8thsign, 2018-08-12:
When a USB Drive is mounted, there should be a check to see if the correct folder structure is there and if it is not, create all the necessary folders as seen in https://github.com/toolboc/psx-pi-smbshare/blob/master/Assets/smbshare.PNG

toolboc, 2018-08-12:
As-is, the folder structure will be created for you when OPL successfully connects to the mounted USB drive for the first time (assuming there is no write protection on the drive). 

Can you confirn whether the above behavior is working as intended?  If it isn't I'm happy to investigate a fix later this week.

Thank you!

Clydefrosch22, 2018-08-12:
Slight suspicion, but what's your storage file system? (Like fat32, ntfs etc)

the8thsign, 2018-08-13:
Yes, OPL did make the correct folders once I connected. Haven't connected to Multiman so I can't say for sure but I'm sure it would work also.

toolboc, 2018-08-13:
Excellent! 

Apologies for this not being obvious. 

Do you have any suggestions to make this more obvious to other users?  I am worried that if it waa an issue to you that it may also be an issue to others.

I would be happy to accept a PR that explains this behavior clearly in the README if you think it would help others.

Let me know what you think, then we can probably close the issue.

Thanks again!

toolboc, 2018-08-13:
*closed*

Functionality already provided when connecting to OPL / Multiman for first time.

closed 2018-08-13 by toolboc

ITotalJustice, 2019-06-01:
Just to add, when connecting a ntfs hdd, no folders are created even when connecting to multiman. The hdd is succesfully mounted to /media and shows in windows when doing \\SMBSHARE\share. However the ps3 still uses the /share folder thats on the sd card. So dumping games to the server simply saves it to the sd card. And when creating all the folders manually, still the /share sd card storage is used instead of the ntfs hdd.

Not sure if you want me to open a new issue on this?

toolboc, 2019-06-02:
@ITotalJustice, thank you for reporting the issue, it looks like the issue you are describing snuck in the previous release when I performed an update to the underlying Raspbian base image which caused some issues with pmount that went unnoticed.  I was able to resolve the issue by re-running setup.sh which in effect rebuilt pmount from source and linked with the updated system modules.

I can confirm that NTFS drives are now working as expected and that Multiman will create the appropriate folder paths when connected (PSXISO and PS2ISO will create upon ripping a respective disc):

[image]

To apply these fixes you can update an existing installation by running the setup.sh located in the home directory via an ssh session, or reflash your SD card with the latest v1.7 release. 

Apologies for any inconvenience this blip may have caused!

ITotalJustice, 2019-06-25:
> @ITotalJustice, thank you for reporting the issue, it looks like the issue you are describing snuck in the previous release when I performed an update to the underlying Raspbian base image which [...]

I completely missed your reply (and quick fix!) until now! Thank you very much