toolboc/psx-pi-smbshare #10: Raspberry pi 4 update?

VersatileNinja, 2019-10-05:
Can you update the image to be compatible with raspberry pi 4? I have the 2Gb model and it just black screens with the current image.

toolboc, 2019-10-05:
@VersatileNinja,

https://raw.githubusercontent.com/toolboc/psx-pi-smbshare/master/setup.sh

I do not have a Raspberry Pi4 on hand to test, however, the [setup script](https://github.com/toolboc/psx-pi-smbshare/blob/master/setup.sh) in this project should be capable of bootstrapping any device running a modern Debian based OS like Raspbian or Ubuntu. 

To do this:
* Download and flash latest [Raspbian Lite image](https://www.raspberrypi.org/downloads/raspbian/)
* Boot and enable SSH on your device
* SSH into the device using default "user=pi , password=raspberry"
* Run the following: 
```
cd ~
wget -O setup.sh https://raw.githubusercontent.com/toolboc/psx-pi-smbshare/master/setup.sh
chmod 755 setup.sh
sudo ./setup.sh
```
* The device will reboot and should wake up as a fully operational psx-pi-smbshare

Let me know if you encounter any issues, it should work^TM

jamesfarnham, 2019-10-09:
I get;

sudo: /etc/init.d/samba: command not found
DAEMON_CONF="/etc/hostapd/hostapd.conf"
interface=wlan1
#driver=nl80211
ssid=XlinkKai
hw_mode=g
channel=7
#ieee80211n=1
wmm_enabled=0
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=XlinkKai
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
interface wlan1
static ip_address=10.254.0.1/16
nohook wpa_supplicant

interface=wlan1
bind-dynamic
dhcp-range=10.254.0.2,10.254.0.100,12h
Failed to start hostapd.service: Unit hostapd.service is masked.
Config directory: /root/.xlink/
Initializing XLink Kai engine...

terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  bind: Address already in use
./setup.sh: line 75:  5400 Aborted                 sudo kaiengine_arm
Connection to 192.168.10.19 closed by remote host.

At the end of the installation. Full log here: https://pastebin.com/Eg3eU7MB

Is there any way to fix it? Thanks!

toolboc, 2019-10-09:
@VersatileNinja.

Looking at the logs it looks like you ran the setup script more than once, which is fine but is the reason you are seeing this:

>terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injectorboost::system::system_error >'
what(): bind: Address already in use

This is because XLink Kai was already running at the time of execution and can basically be ignored.

Aside from that, logs look good with exception of 

>sudo: /etc/init.d/samba: command not found

I assume this is due to samba now being called smbd on latest Raspbian?  I'm not entirely sure because I don't have a pi4 to test. You can verify by list the content of /etc/init.d. If that is the case, changes would need to be made to the [following](https://github.com/toolboc/psx-pi-smbshare/search?q=init.d&unscoped_q=init.d).

I've stood up a branch which changes all calls to /etc/init.d/samba to /etc/init.d/smbd.  Using your pi4 as-is, you should be able to try it out with (of course ignore the Xlink Kai A:

jamesfarnham, 2019-10-10:
Here's another attempt on a fresh install, installing over the top had the same effect I think. It gets up to "Kai Engine: The engine is rocking
Connect to: http://127.0.0.1:34522/", but can't connect to samba shares.

Thanks for your help!

https://pastebin.com/DzixUSma

toolboc, 2019-10-10:
@jamesfarnham,

Try installing using the smbd branch, according to the logs it looks like you used master (No need to fresh install, just rerun on top of existing):

```
cd ~
wget -O setup.sh https://raw.githubusercontent.com/toolboc/psx-pi-smbshare/smbd/setup.sh
chmod 755 setup.sh
sudo ./setup.sh
```

Thank you for testing!

jamesfarnham, 2019-10-10:
Thanks! Now I get:

https://pastebin.com/cZiD5gYV

Which looks fine ... But I'm not able to log in to the Samba server over Wi-fi. It throws up a login error in each case logging in with guest, or pi:raspberry. Is there a Samba log I could send that would help narrow that down?

I can see the share on Mac/Iphone file sharing app, but not log into it.

jamesfarnham, 2019-10-10:
Ps2 over Ethernet gets the same error; 300 cant connect to SMB

jamesfarnham, 2019-10-10:
Actually, running ./samba-init.sh got it to complete the mount on my mac; 

pi@raspberrypi:~ $ ./samba-init.sh
[global]
workgroup = WORKGROUP
usershare allow guests = yes
map to guest = bad user
allow insecure wide links = yes
[share]
Comment = Pi shared folder
Path = /share
Browseable = yes
Writeable = Yes
only guest = no
create mask = 0777
directory mask = 0777
Public = yes
Guest ok = yes
force user = pi
follow symlinks = yes
wide links = yes

Then I could edit the /etc/samba/smb.conf file to change the path from /share to /media/sda1 - I guess that was the problem not letting the connection happen. I restarted samba and it works!

toolboc, 2019-10-10:
@jamesfarnham,

Just FYI, when you plug in your external media (sda1), the automount-usb.sh script should automatically mount /sda1 to /share.  Would you be able to verify this?  I hope automount isn't broken on the pi4.

Cheers!

jamesfarnham, 2019-10-11:
smb.conf before plugging in usb:

[global]
workgroup = WORKGROUP
usershare allow guests = yes
map to guest = bad user
allow insecure wide links = yes
[share]
Comment = Pi shared folder
Path = /share
Browseable = yes
Writeable = Yes
only guest = no
create mask = 0777
directory mask = 0777
Public = yes
Guest ok = yes
force user = pi
follow symlinks = yes
wide links = yes

media empty

after plugging in usb:

[global]
workgroup = WORKGROUP
usershare allow guests = yes
map to guest = bad user
allow insecure wide links = yes
[share]
Comment = Pi default shared folder
Path = /media/sda5
Browseable = yes
Writeable = Yes
only guest = no
create mask = 0777
directory mask = 0777
Public = yes
Guest ok = yes
force user = pi
follow symlinks = yes
wide links = yes

pi@smbshare:/media $ ls
sda1  sda2  sda3

I only had to change sda5 to sda1 in the .conf and restart Samba to make it work this time. Before I wasn't plugging in the USB live, only booting with it already plugged in, if that could make a difference.

Always willing to break something in the name of science! Thanks again for your help and support.

toolboc, 2019-10-11:
@jamesfarnham,

Hmmm, it seems automount is working, the drive letter being (sda5) may or may not be an issue depending on what value was assigned at time of insertion.  It should also work whether the drive is hot plugged or already inserted at boot time.  It looks like this is mostly working though.

I’ll try to source a pi4 in the meantime and close this out when I can confirm it locally.

VersatileNinja, 2020-02-23:
Just wanted to give a quick update as I revisited this recently with a 1GB RPI4.

After using the latest Raspbian lite (Feb 2020 release), I ran the following script:

```
cd ~
wget -O setup.sh https://raw.githubusercontent.com/toolboc/psx-pi-smbshare/master/setup.sh
chmod 755 setup.sh
sudo ./setup.sh
```

Let it do its thing, and then I reboot at the end for safe measure. I insert a usb drive (usb 2.0 or usb 3.0 port on RPI4 doesn't matter) and PS2 recognizes the smb share.

I assume xlink kai is working but I haven't tested that yet on my 360.

toolboc, 2020-02-23:
Thank you for verifying, I’ve still not acquired a pi4 of my own so definitely appreciate your help here.  

I’m going to keep the issue open until I get a device of my own and cut a release for pi4.

If any issues are noted, please let us know here.

Thanks again!

VersatileNinja, 2020-02-24:
I can confirm I can access xlink kai on my rpi4 via the web browser by using pi_ip_address:34522

However, I cannot access the mounted usb share through windows. HTTP:\\smbshare\share doesn't work like the readme webpage. In fact, I can't even the folder shared under Windows 10 under Windows Explorer under Network icon.

It's not that big of a deal, but if I could figure out how to access the share that be awesome thru Windows that be awesome.

toolboc, 2020-04-09:
@VersatileNinja, it looks like a recent Windows 10 update now disables access to insecure Samba shares.  For now, you can enable insecure Samba access on your PC by following these instructions: https://devanswers.co/network-error-problem-windows-cannot-access-hostname-samba/

I will look into enabling a user/pass by default in future release to accommodate this.

closed 2020-04-09 by toolboc