toolboc/psx-pi-smbshare #13: How to connect to web interface of game managers?

ectorhga, 2019-12-23:
Hi,
usually I just have to enter the IP of the PS3 to connect to the web interface of the game managers (webman Mod or Sman). Now, the PS3 is behind the router and seem to be not accessible. 
What do I have to do to reach the web interface of the game managers mentioned above?

toolboc, 2019-12-29:
I’m not entirely familiar with your use case, but you should also be able to access the services you have mentioned from the PS3’s built-in browser, without any additional steps it should just work.

However, if you need to access these services from a different device, one way to accomplish this is to plug in a second WiFi dongle that is compatible with your raspberry pi.  This will serve up an insecure network from your device named “XlinkKai”.  Connect to this network from your client device and you should be able to access any services running on your system using the default assigned ip of 192.168.2.2

For example, if a web service happens to be running on port 80 of your PS3, after connecting to the “XlinkKai” network, visit http://192.168.2.2 in a web browser to access the web service.

ectorhga, 2020-01-08:
Hi, thanks for the explanation! I'll try it this way!

closed 2020-01-08 by ectorhga