toolboc/psx-pi-smbshare #14: not an issue just a question related with xlink kai and EW-7811Un

alfredox123, 2020-01-12:
i see that on the psp guide for xlink kai have been tested with a usb wifi adapter model EW-7811Un, i wanted to ask if this same adapter could work for linking psp to the pc with the pc version of xlink kai, im asking here cuz i dont know where else to ask it. thanks for your time

toolboc, 2020-01-12:
@alfredox123,

The EW-7811Un is also the model I used when testing the PSP functionality so it should be the most reliable adapter choice for connecting a PSP to XlinkKai in psx-pi-smbshare.

Cheers,

Paul

closed 2020-01-12 by toolboc

alfredox123, 2020-01-12:
i see that it works well on psx-pi-smbshare but im asking for the PC version of xlink kai, is it the same regarding psp link connectivity? or do i have to find old adapters recomended on the xlink kai wiki? which seems pretty outdated list

toolboc, 2020-01-12:
@alfredox123,

Provided that you have appropriate drivers installed, it “should” work on PC, I cannot confirm by experience though.  The key component is that the WiFi dongle needs to be capable of supporting ad-hoc mode which is possible with the EW-7811Un.

-Paul