toolboc/psx-pi-smbshare #15: "Upgrading WEB UI" loop

domeus, 2020-04-03:
WEB GUI just looping "UPGRADING WEB UI" and newer ends. Any tips?

aapokaapo, 2020-04-09:
Delete the webui.zip in /home/user/.xlink/

domeus, 2020-04-09:
> Delete the webui.zip in /home/user/.xlink/

Thanks, now webui returns "404 Page Not Found" I removed file from  /root/.xlink/webui.zip because Im using raspberry pi image version

aapokaapo, 2020-04-09:
After deleting that zip, you need to restart xlink. It should autoupdate and download the latest webui.zip

domeus, 2020-04-09:
> After deleting that zip, you need to restart xlink. It should autoupdate and download the latest webui.zip

Rebooted, still looping :(

aapokaapo, 2020-04-09:
Interesting, do you mind joining the xlink discord so we coukd troubleshoot this further https://discord.gg/aResBT

domeus, 2020-04-09:
> Interesting, do you mind joining the xlink discord so we could troubleshoot this further https://discord.gg/aResBT

Just joined, looks like someone else have same issue

toolboc, 2020-04-09:
It looks like there have been some changes that require updating the XlinkKai client, this has been addressed in https://github.com/toolboc/psx-pi-smbshare/commit/1a584a182bd7e767d81d6309d17e061dbe4d8850

closed 2020-04-09 by toolboc