toolboc/psx-pi-smbshare #21: how to change game for psp online with xlink kai?

alfredox123, 2020-07-23:
when i change the game, the only way i mange to connect my pi with my vita is rebooting the whole rpi, is there a command that im missing to shutdown connection between vita and the pi?

also wanted to let you know that asus n13 usb adapter also works with this, THANK YOU VERY MUCH!

toolboc, 2020-07-23:
@alfredox123,

Very interesting, I never tried using XLinkKai with the PS Vita, but would be very thankful if you might be able to contribute steps for using this device in the official README.   You should be able to shutdown the connection between the PS Vita and pi with:
```sudo iw dev wlan0 disconnect``` 

See: http://ict.siit.tu.ac.th/help/iw for more details /options

alfredox123, 2020-07-23:
later i will try this thx, just for more info, im using a psp game on my ps vita. i just followed your guide and it worked but i think you need to add to that guide that you need to rename wifi-to-eth-route.sh (to rename wifi-to-eth-route.sh1 for example) IF you have your pi connected through ethernet and using a psp xlink mode capable usb adapter. took me a few hours to discover that lol

toolboc, 2020-07-23:
@alfredox123 , thank you for the suggestion, the documentation has been updated in https://github.com/toolboc/psx-pi-smbshare/commit/ec6eb0fe096990d183b543edb7e34e33271a6332

alfredox123, 2020-07-25:
im still getting command failed: operation already in progress (-114) when trying to connect to a new game with command sudo iw wlan1 ibss join <SSID> <frequency> (used the command sudo iw dev wlan0 disconnect before connecting to the new game)

toolboc, 2020-07-25:
This may be an issue with XlinkKai, what happens if you try forcing the XlinkKai engine to restart with the following:
```
sudo pkill kaiengine
```

alfredox123, 2020-07-26:
same command failed

toolboc, 2020-07-26:
I appreciate the assistance, unfortunately this isn't a use case I'm currently equipped to replicate. You could also try to restart the wireless interface with (assumes wlan0 is the interface that the PSP is connecting to):

```
ifdown wlan0
ifup wlan0
```

closed 2020-08-01 by toolboc

alfredox123, 2020-08-08:
sorry for late response "sudo iw dev wlan1 del" gives me command failed unknown error 524 (-524)

alfredox123, 2020-08-08:
nvm sudo ifconfig wlan1 down and sudo ifconfig wlan1 up did the trick thx!

alfredox123, 2020-09-12:
just to clarify, you dont need to rename wifi-to-eth-route.sh for playing psp online, just tested the wireless method (wlan0 "internal wifi" - wireless internet and wlan1 "usb adapter" - psp connected to rpi3) and it works to