toolboc/psx-pi-smbshare #22: (no an issue) posible to transfer games to internal hdd on og xbox?

alfredox123, 2020-07-23:
is it posible to transfer games that i have on my thumbdrive, connect it to the pi, connect xbox with ethernet cable to the same router, then remote control pi to transfer the games from thumbdrive to internal hdd of the og xbox?

toolboc, 2020-07-23:
This is probably possible using a dashboard that supports [SMB fileshares on the OG Xbox](https://www.xbmc4xbox.org.uk/wiki/Windows_File_Sharing_(SMB/SAMBA/CIFS)) (for example [XBMC4Xbox](https://www.xbmc4xbox.org.uk/)).  By default, [psx-pi-smbshare will auto-mount and serve USB drives via SMB](https://github.com/toolboc/psx-pi-smbshare#accessing-usb-drives-on-the-smb-share) to `\\SMBSHARE\share`. 

In the XBMC dashboard, you could try to [configure a new SMB source for this path](https://www.xbmc4xbox.org.uk/wiki/Configuring_PC_Shares), then copy files to the OG Xbox HD using the [filemanager](https://www.xbmc4xbox.org.uk/wiki/Filemanager).

closed 2020-07-23 by toolboc

alfredox123, 2020-08-01:
do i really need a router to make this happen? with a crossover cable cane be done?

toolboc, 2020-08-01:
A standard ethernet cable between the Xbox and the pi can provide a direct connection to the SMB share without the need for a router.

alfredox123, 2020-08-01:
i just tried it but it didnt connect, it worked connecting it with a router though. i dont know if renaming wifi-to-eth-route.sh has something to do?

toolboc, 2020-08-01:
If you renamed wifi-to-eth-route.sh, you have effectively disabled the DHCP service that normally runs on the pi.  The DHCP service must be enabled on the pi if you wish to access the SMB share directly using an ethernet cable connecting the console to the pi.