toolboc/psx-pi-smbshare #23: Is exfat usb automatic mount supported?

VersatileNinja, 2020-08-13:
I was going to do a quick usb test with my pi4 and I had a game on a external 256gb usb 3.0 thumb drive, formatted at exfat.

In OPL, the SMB share error was 303.

I used a usb flash drive 2.0 that was 8gb, formatted as fat32, and all games were detected fine, no smb share error.

How do I use a exfat usb thumb drive for ps2 use and smb share?

Likewise, is NTFS supported? The readme file on main github doesn't go over these exact details.

toolboc, 2020-08-13:
The [automount-usb.sh](https://github.com/toolboc/psx-pi-smbshare/blob/master/automount-usb.sh#L17) script enables support for exfat and NTFS drives.  It has been noted that exfat drives may require GPT based partitioning and may not work if it is configured as MBR, see this comment for more details: https://github.com/toolboc/psx-pi-smbshare/issues/20#issuecomment-660396120

VersatileNinja, 2020-08-13:
Thanks for clarifying. I use windows 10 so I will look into how to make a GPT based partitioning and I should be good to go based upon comments you linked.

closed 2020-08-13 by VersatileNinja