toolboc/psx-pi-smbshare #27: transfer from usb on rpi to xbox not working

alfredox123, 2020-10-25:
today i tried transfer a game to og xbox and it didnt work, it said something about couldnt find name of smb or something among those lines, however if i disconnect the usb it shows what it is on the sd card of the rpi. tried putting usb on ntfs and fat32 and neither worked

toolboc, 2020-10-26:
Does the USB drive mount properly to `/share`?  If the drive mounts properly, it should be usable as an SMB source.  You can verify this by navigating to `//smbshare/share` after before and after plugging in the USB drive, you mentioned that the SD card contents mount but not the drive, this would suggest that you may have an issue with the USB drive. If you are having issues mounting a particular drive, you could try changing the partition table to / from MBR / exFat as mentioned here: https://github.com/toolboc/psx-pi-smbshare/issues/20#issuecomment-660396120

alfredox123, 2020-10-26:
it was working well before, usb works as it should connecting it to my pc, i did notice that i have to leave it on and connected for several minutes to actually acces atleast the sd card from og xbox (no problems doing scp from my pc), later i will try the exfat format

closed 2020-11-17 by toolboc