toolboc/psx-pi-smbshare #28: (not an issue) automated/simpler process to play online psp games?

alfredox123, 2020-11-20:
hello its me again, is there a simpler way to executed all this commands

sudo ifconfig wlan1 down
sudo ifconfig wlan1 up
sudo service hostapd stop
sudo iw wlan1 set type ibss
sudo iw wlan1 ibss join <SSID> <frequency>

without put in one by one?, some thing like making a file with all that commands and then put in command line something like "execute cod.txt" and it makes all those lines for me without writing it one by one?

toolboc, 2020-11-21:
These commands can probably be executed in a custom shell script.  If you are new to Linux in general, there is a great guide available [here](https://linuxconfig.org/bash-scripting-tutorial-for-beginners).  I might be interested in accepting a pull request if someone can create and test a working script for this.

alfredox123, 2020-12-17:
sorry for late response, thank you very much for that guide, helped me alot, now i can play psp games online without being so painfull lol

closed 2020-12-17 by alfredox123