toolboc/psx-pi-smbshare #29: something seems to be broken with last update maybe, with psp and xlink kai online

alfredox123, 2020-12-18:
just updated to your latest version and it takes me a few reboots of my rpi to connect with wifi signal, has been hapening since i updated it.

also, when i finally connect to my wifi i do all the steps to try play online with xlink but for some reason doesnt show that im broadcasting output, when i boot the game it broadcast cast signal for 1 or 2 seconds, but after that it doesnt broadscast signal anymore and for that reason i cannot find matches nor host with xlink kai.

dont know if this is a software or hardware related but it started happening after i update it.

toolboc, 2020-12-18:
Have you tried reverting to a previous version to see if these issues persist?  That could help narrow down a root cause (hardware / software/ some combination?).

alfredox123, 2020-12-18:
if i go to a previous version xlink kai will be outdated right? which means i couldnt connect with other users :S

toolboc, 2020-12-18:
XlinkKai does not necessarily require that two clients are on the same version.  However, to rule that out you could independently upgrade XlinkKai after reverting to a previous release of psx-pi-smbshare. 

So far, this issue has not zeroed in on a root cause.  Given that you are experiencing intermittent network issues, it is advised to downgrade to see if the issue resolves itself, if the issues continue you may want to check if there is an issue with your WiFi / network hardware as this appears to be an isolated issue at this time.

alfredox123, 2020-12-18:
how to update xlink kai indepently?

toolboc, 2020-12-18:
psx-pi-smbshare uses the debian package source for XlinkKai as of version 1.8, for a quick overview of the changes that accommodate this you can refer to commit https://github.com/toolboc/psx-pi-smbshare/commit/893a893cc473ee390e21fdaf9b91029f2ca70ae5.

To update XlinkKai independenty, you can run the relevant portions of that changeset by copying and pasting the following into a terminal session:

```# Install Xlink Kai
wget -q -O - https://repo.teamxlink.co.uk/debian/KEY.gpg | sudo apt-key add -
sudo echo 'deb https://repo.teamxlink.co.uk/debian/ /' > /etc/apt/sources.list.d/teamxlink.list
sudo apt-get update
sudo apt-get install xlinkkai

cat <<'EOF' > /home/pi/launchkai.sh
while true; do
    /usr/sbin/kaiengine
    sleep 1
done
EOF

chmod 755 /home/pi/launchkai.sh
```

After running this, you should be able to update XlinkKai at any time with: `apt-get install xlinkkait`

alfredox123, 2020-12-18:
i see that i have an old iso stored must be v1.7 or v1.8 anyway to check exactly which version is?

toolboc, 2020-12-18:
You could check the date of your image file against the release dates listed [here](https://github.com/toolboc/psx-pi-smbshare/tags).  For quality control, it is advised to re-download a specific version and go from there.

alfredox123, 2020-12-19:
seems like a hardware issue, just tested with v1.9 and couldnt connect with wifi no matter what, but i can with ethernet.

with ethernet connected i tested if i do broadcasting output and the same, it only broadcast at boot of the game but not anymore...

seems like both rpi and dongle for psp are dying...

closed 2020-12-19 by toolboc

alfredox123, 2020-12-20:
i did some more testing and dongle for psp works as it should on windows xp, gonna keep trying a few things

S4bg0, 2023-04-18:
> psx-pi-smbshare uses the debian package source for XlinkKai as of version 1.8, for a quick overview of the changes that accommodate this you can refer to commit [893a893](https://github.com/toolboc/psx-pi-smbshare/commit/893a893cc473ee390e21fdaf9b91029f2ca70ae5).
> 
> To update XlinkKai independenty, you can run the relevant portions of that changeset by copying and pasting the following into a terminal session:
> 
> ```
> wget -q -O - https://repo.teamxlink.co.uk/debian/KEY.gpg | sudo apt-key add -
> sudo echo 'deb https://repo.teamxlink.co.uk/debian/ /' > /etc/apt/sources.list.d/teamxlink.list
> sudo apt-get update
> sudo apt-get install xlinkkai
> 
> cat <<'EOF' > /home/pi/launchkai.sh
> while true; do
>     /usr/sbin/kaiengine
>     sleep 1
> done
> EOF
> 
> chmod 755 /home/pi/launchkai.sh
> ```
> 
> After running this, you should be able to update XlinkKai at any time with: `apt-get install xlinkkai`

Just to let people know that repo url has recently changed and is now located at https://dist.teamxlink.co.uk/.
Can help if you want to update.