toolboc/psx-pi-smbshare #30: [question] Is it possible to play games hosted on the Pi via OPL through XLink Kai?

cdlenfert, 2020-12-20:
I've converted all my PS2 games to ISOs for OPL, but in order to play over SMB, my PS2 (192.168.0.10) has to be on the same network as my samba server (192.168.0.8). With your image, can the Pi serve the games via SMB AND host the xlink server? I'm thinking it could work by hooking the PS2 ethernet directly to the Pi, and having the Pi on wifi. If that's the whole point of this, I apologize for the silly question. I'd just like to confirm it's possible before making a big change to my setup. Thanks!

toolboc, 2020-12-20:
>Is it possible to play games hosted on the Pi via OPL through XLink Kai?  With your image, can the Pi serve the games via SMB AND host the xlink server? I'm thinking it could work by hooking the PS2 ethernet directly to the Pi, and having the Pi on wifi 

This is not possible due to OPL's SMB support requiring exclusive access to the PS2's network adapter (meaning that once you commit to streaming a game over SMB, the network adapter will not be available to the game being streamed).   

That said, this project does host the Samba and XlinkKai services on the pi itself, and repurposes the ethernet port of the pi to act as a fast throughput channel for accessing those services.  In this manner, games are steamed directly through the ethernet port from the Pi to the PS2, while XlinkKai communications are transmitted over Wi-Fi, however the above limitation still exists and you are effectively only able to use of one of these services at a time in practice when connected to a Playstation2.  This diagram describes the current functionality of psx-pi-smbshare:

[image]

cdlenfert, 2020-12-20:
Thank you for taking the time to provide such a thorough response. I didn't think it would be possible to do SMB games and XLink Kai at the same time, so I appreciate the confirmation.

closed 2020-12-20 by cdlenfert