toolboc/psx-pi-smbshare #31: psx-pi-smbshare on lakka

kieuphong, 2021-02-15:
Hi, 

Can you make this work on top of lakka on rpi3?  I don't want to waste a pi 3 just to use smbshare.  Thanks!

KP

toolboc, 2021-02-15:
@kieuphong,

This will probably work by running the commands mentioned [here](https://github.com/toolboc/psx-pi-smbshare#upgrading-an-existing-install) on top of your existing installation:

> cd ~
> wget -O setup.sh https://raw.githubusercontent.com/toolboc/psx-pi-smbshare/master/setup.sh
> chmod 755 setup.sh
> sudo ./setup.sh

kieuphong, 2021-02-15:
I did that and got this:

Lakka (official): 2.3.2 (RPi2.arm)
Lakka:~ # cd ~
Lakka:~ # ls
assets       database     overlays     roms         screenshots  system
backup       joypads      playlists    savefiles    setup.sh     thumbnails
cores        lost+found   remappings   savestates   shaders
Lakka:~ # file setup.sh
-sh: file: not found
Lakka:~ # cd ~
Lakka:~ # wget -O setup.sh https://raw.githubusercontent.com/toolboc/psx-pi-smbs
hare/master/setup.sh
Connecting to raw.githubusercontent.com (185.199.109.133:443)
setup.sh             100% |*******************************|  3004   0:00:00 ETA
Lakka:~ # chmod 755 setup.sh
Lakka:~ # nano setup.sh
Lakka:~ # sudo ./setup.sh

 There is no working 'sudo'.

 On debian/ubuntu/all general purpose linux distributions 'sudo'
 allows a permitted user to execute a command as the superuser
 or another user, as specified by the security policy

 With LibreELEC you have root access by default, so you dont need 'sudo'

Lakka:~ # ./setup.sh

 There is no working 'sudo'.

 On debian/ubuntu/all general purpose linux distributions 'sudo'
 allows a permitted user to execute a command as the superuser
 or another user, as specified by the security policy

 With LibreELEC you have root access by default, so you dont need 'sudo'

 There is no working 'sudo'.

 On debian/ubuntu/all general purpose linux distributions 'sudo'
 allows a permitted user to execute a command as the superuser
 or another user, as specified by the security policy

 With LibreELEC you have root access by default, so you dont need 'sudo'

 There is no working 'sudo'.

 On debian/ubuntu/all general purpose linux distributions 'sudo'
 allows a permitted user to execute a command as the superuser
 or another user, as specified by the security policy

 With LibreELEC you have root access by default, so you dont need 'sudo'

Connecting to raw.githubusercontent.com (185.199.111.133:443)
wget: can't open '/home/pi/samba-init.sh': No such file or directory
chmod: /home/pi/samba-init.sh: No such file or directory

 There is no working 'sudo'.

 On debian/ubuntu/all general purpose linux distributions 'sudo'
 allows a permitted user to execute a command as the superuser
 or another user, as specified by the security policy

 With LibreELEC you have root access by default, so you dont need 'sudo'

 There is no working 'sudo'.

 On debian/ubuntu/all general purpose linux distributions 'sudo'
 allows a permitted user to execute a command as the superuser
 or another user, as specified by the security policy
[... 114 lines cut ...]
 There is no working 'sudo'.

 On debian/ubuntu/all general purpose linux distributions 'sudo'
 allows a permitted user to execute a command as the superuser
 or another user, as specified by the security policy

 With LibreELEC you have root access by default, so you dont need 'sudo'

./setup.sh: line 75: ps3netsrv++: not found

 There is no working 'sudo'.

 On debian/ubuntu/all general purpose linux distributions 'sudo'
 allows a permitted user to execute a command as the superuser
 or another user, as specified by the security policy

 With LibreELEC you have root access by default, so you dont need 'sudo'

 There is no working 'sudo'.

 On debian/ubuntu/all general purpose linux distributions 'sudo'
 allows a permitted user to execute a command as the superuser
 or another user, as specified by the security policy

 With LibreELEC you have root access by default, so you dont need 'sudo'

toolboc, 2021-02-15:
Looks like your distro does not support "sudo" because it runs as root by default.  

You can either:
A.) Modify the setup.sh script and all supporting scripts to remove all instances of "sudo" and hope that it works
B.) Use another SD card instead of trying to do everything in a single OS image, this would leave you with a supported installation that is free of any potential issues and allow you to only need 1 Raspberry Pi device (just swap the SDs as needed).

closed 2021-02-15 by toolboc