toolboc/psx-pi-smbshare #32: OPL can see the share, but won't list files.

BOBdotEXE, 2021-04-28:
I've tried using the latest OP: 1.0.0, as well as the latest nightly.
I'm using the provided image, on a raspberry pi 3, and the only thing I've done is update the installed packages.

I've double-checked my network settings, and everything seems fine,
The pi has wifi access, and is connected via Ethernet to the ps2 (slim)

If I delete the share name, and re-connect, OPL can see that there's a share called 'share' and allows me to open it.
Once I open it, there are no games listed.

I've double-checked via putty, and connected to the share in windows, the USB drive has been mounted properly.
(I see the files in \\192.168.0.225\share\DVD Both the ISO and Games.bin)

I've even tried unmounting the Drive, and placing ISO's directly in /share/DVD on the rpi.

Same result, (I removed the files from /share/dvd afterwards)
I tried adjusting the samba settings, thinking OPL's version was too new, and added:
max protocol = NT1
min protocol = CORE
ntlm auth = yes

Then restarted samba, No change,
It also looks like samba resets my settings on reboot.

I've also tried using the exact same USB drive, plugged directly into the PS2, and it was able to read the files just fine.
So I don't think it's a directory/ file name issue.

I've also tried manually entering the username/password for the pi in OPL network settings.
No luck

Any ideas?

toolboc, 2021-04-28:
I wonder if this could be due to changes in the latest OPL, are you able to verify that it works with any earlier version of OPL?

BOBdotEXE, 2021-04-28:
> I wonder if this could be due to changes in the latest OPL, are you able to verify that it works with any earlier version of OPL?

Sadly, I've deleted my older copies, but if you have a version recommendation, I'll download that.

closed 2021-04-28 by BOBdotEXE