toolboc/psx-pi-smbshare #33: [Minor issue] Root owns files I wish to edit for adding a power off button to PI 4B

Nikhchan, 2021-06-18:
So I have the PSX Pi smbshare running successfully on Pi 4b using the methods discussed [here](https://github.com/toolboc/psx-pi-smbshare/issues/10).

But I am trying to add a power off button via GPIO3 pin. For that, I need to add following command in the boot/config.txt file. 

`#shutdown button
dtoverlay=gpio-shutdown`

But the file seems to be only writable by 'root' and everyone else can only read and execute. 

Now I am not aware of this root user password. I tried the usual defaults like root, toor, raspberry, strawberry, etc. None worked..

Anyway to edit that file or if you can share that root's pw, that would be great. I have backed up the default config.txt file.  

Thank you for making this great tool!

toolboc, 2021-06-18:
psx-pi-smbshare is built on top of stock Raspbian, as such there is no “root” user.  There is a default user “pi” with a password of “raspberry” that has super user permissions.  You can therefore edit the file in question by ssh’ing into the device as the ”pi” user and running ‘sudo vi /boot/config.txt’.

Nikhchan, 2021-06-19:
> psx-pi-smbshare is built on top of stock Raspbian, as such there is no “root” user. There is a default user “pi” with a password of “raspberry” that has super user permissions. You can therefore [...]
Thanks for the reply. 
I know. but the bootconfig.txt file had 'root' as the owner. and it wouldn't allow me to modify it through the Pi user. Neither through ssh nor through directly via text editor. I thought it was something to protect the functionality of the various features of psx pi smbshare.

Thankfully I have rectified the issue. And its clearly not related to the psx-pi-share.

closed 2021-06-19 by toolboc