toolboc/psx-pi-smbshare #37: Add Raspberry Pi 4 Support/image

noisenstatic, 2021-12-07:
First, thank you so much for this tool! 

Now is it possible to add support to Pi4? I'm having a hard time setting it up. I saw the link stating some of the commands but on the latest version of linux it doesn't seem to boot the smb share.

Please let me know if I can help testing things out, I recently got a Pi4 2gb of ram just to do this. I'm using multiple PS3 and the gigabit ethernet + USB3 would go a long way.

noisenstatic, 2021-12-11:
Finally found the issue. The problem was that exFAT format is not being properly mounted while using rasp4 +manual setup.  Migrated to NTFS for my external drive and it's working fine now.

VersatileNinja, 2022-01-05:
> Finally found the issue. The problem was that exFAT format is not being properly mounted while using rasp4 +manual setup. Migrated to NTFS for my external drive and it's working fine now.

Can I ask you a question. I have a pi4 4gb and I ran the script in terminal. Seems to install fine. Use a fat32 USB and it works with opl on PS2, but when I plug into the pi4, it's not automount the usb contents to the smb share folder.

Not sure why because I remember doing this like over a year ago on older version of raspbian with no issues. If you got this working recently over smb, please share any tips.