toolboc/psx-pi-smbshare #40: Found a fix to make psx-pi-smbshare work with pi4 and latest 64-bit raspberry pi os

VersatileNinja, 2022-02-05:
Recently, I installed the 64-bit version of raspberry pi os (with desktop interface) to my pi 4, 4GB model.

I ran the below script:

`cd ~
wget -O setup.sh https://raw.githubusercontent.com/toolboc/psx-pi-smbshare/master/setup.sh
chmod 755 setup.sh
sudo ./setup.sh`

Followed the main page to setup the PS2 OPL network settings, and it wouldn't connect to the pi 4! I knew it did in the past.

After some more research on google and trying things out hit or miss, I found the solution.

Do the following:

`sudo nano /etc/samba/smb.conf`

After that, add the following code under [GLOBAL] section:
`min protocol = NT1`

Save the file, and then restart samba service by:
`sudo systemctl restart smbd`

Now go back to your OPL and it should work! Hopefully toolboc sees this and implements into the master!

toolboc, 2022-02-06:
@VersatileNinja, awesome work and thank you for sharing the solution!  I think this is a good change to implement going forward as Windows is phasing into a more secure-by-default default settings for SMB in general.  This should be legacy enough for working with OPL and also modern enough to work with most other OSs.  I’ll need to find time to test this on more setups but for now this a great solution / workaround!

closed 2022-02-06 by toolboc

VersatileNinja, 2022-02-06:
> @VersatileNinja, awesome work and thank you for sharing the solution! I think this is a good change to implement going forward as Windows is phasing into a more secure-by-default settings for SMB in general. This should be legacy enough for working with OPL and also modern enough to work with most other OSs. I’ll need to find time to test this on more setups but for now this a great solution / workaround!

-------

It be helpful if other users can test this out. I know when I did a fresh install, and then added this code, it worked great. Now I did a shutdown of the pi, and go back to it later, nothing else changed, and I get a SMB 300 connection error on the PS2. I think maybe some service is not fully started and/or the initial setup.sh script never fully installed itself? It be great if someone more experience with linux can verify. :)