toolboc/psx-pi-smbshare #41: Spiked CPU usage from Kaiengine on Pi Zero W

cdlenfert, 2022-02-10:
When I look at `htop` I see that the `kaiengine` process is using nearly 100% CPU.
[image]

I've has similar issues in the past (not using psx-pi-smbshare) and it was due to the way I had launched the process to run in the background. When I ran the process in the foreground in my shell session, CPU usage was closer to 11%. I was also able to start `kaiengine` in a `screen` session and detach, and usage stayed low.

I realize this may be low on the priority list, but just wanted to share my findings and see if I can assist in coming up with any solutions. I imagine running at 100% CPU over time won't be good for the Pi Zero's longevity.

CrunchBite, 2022-07-22:
Ah its likely TTY related. I'll reproduce this and develop a fix 👍

CrunchBite, 2022-07-23:
Ah its likely TTY related. I'll reproduce this and the team will develop a fix 👍

MasonT8198, 2022-07-26:
This isn't a psx-pi-smbshare issue, this is an XLink Kai issue, we're working on it.

teamxlink-dev, 2023-02-25:
Should be resolved completely by XLink Kai 7.4.44 and the changes in 96a902ec7e10d5b0d326593b73445bb86adfffef

MasonT8198, 2023-02-25:
@cdlenfert please test this and let us know how this works before I close.