toolboc/psx-pi-smbshare #42: Unable to login to xlink, is it oudated?

0xym0ron, 2022-02-15:
Whenever I try to login to my xlink account it just refreshes the same page. Works fine if I run it on my PC.

toolboc, 2022-02-15:
This behavior indicates that the XLinkKai version is likely outdated.  Steps to install the latest version are linked below:

https://github.com/toolboc/psx-pi-smbshare/issues/29#issuecomment-748166998

0xym0ron, 2022-02-15:
Tried with a fresh install, updating doesn't appear to work.
[image]

Any ideas?

teamxlink-packaging, 2022-09-04:
Fixed in https://github.com/toolboc/psx-pi-smbshare/pull/48

AgentTechnoman, 2022-10-21:
For me, the fix was making sure my xlink password didn't contain any characters that are interpreted by BASH. YMMV.