toolboc/psx-pi-smbshare #50: ./setup.sh stuck at Launching Xlink Kai

Erd4, 2022-09-27:
Hi. 

My Pi has been set up with psx-pi-smbshare but whenever I try to update it and enter:

`cd ~
wget -O setup.sh https://raw.githubusercontent.com/toolboc/psx-pi-smbshare/master/setup.sh
chmod 755 setup.sh
sudo ./setup.sh`

It gets stuck after:
`Job for hostapd.service failed because the control process exited with error code.
See "systemctl status hostapd.service" and "journalctl -xe" for details.
Checking for XLink Kai updates
Reading package lists... Done
Building dependency tree       
Reading state information... Done
xlinkkai is already the newest version (7.4.43-606669993).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Launching XLink Kai`

dstarr5d, 2022-12-12:
I'm having this problem, as well. I install the latest version of Raspberry Pi OS Lite to my RPi4 2GB, sudo apt-get upgrade, sudo apt-get update, sudo apt-get install screen, reboot, then run the script provided, and it gets this far and then just stops. I have to kill the script to be able to do anything else. This is the last bit I see:

```
Failed to start hostapd.service: Unit hostapd.service is masked.
Checking for XLink Kai updates
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
xlinkkai:armhf is already the newest version (7.4.43-606669993).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Launching XLink Kai
```

teamxlink-dev, 2022-12-13:
We'll figure this out and open a PR with a fix 👍

osaeed18, 2022-12-23:
I commented out the following from the https://raw.githubusercontent.com/toolboc/psx-pi-smbshare/master/setup.sh

# Write XLink Kai launch script
#cat <<'EOF' > /home/pi/launchkai.sh
#echo "Checking for XLink Kai updates"
#sudo apt-get install xlinkkai -y
#echo "Launching XLink Kai"
#while true; do
#    screen -dmS kai kaiengine
#    sleep 5
#done
#EOF

#chmod 755 /home/pi/launchkai.sh

The setup completed, rebooted and on the monitor with the raspberry connected, I see Failed to start Access point and authentication server for Wi-Fi and Ethernet. 

pi@raspberrypi:~ $ sudo systemctl status hostapd
● hostapd.service - Access point and authentication server for Wi-Fi and Ethernet
     Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Fri 2022-12-23 00:57:30 EST; 390ms ago
       Docs: man:hostapd(8)
    Process: 2320 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=1/FAILURE)
        CPU: 28ms

I am assuming because I need to attach a secondary wifi hardware to act as a access point? Then it should create a wlan1 interface in ifconfig -a ?

pi@raspberrypi:~ $ journalctl -xe
Dec 23 00:57:55 raspberrypi hostapd[2402]: nl80211: Driver does not support authentication/association or connect commands
Dec 23 00:57:55 raspberrypi hostapd[2402]: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Dec 23 00:57:55 raspberrypi hostapd[2402]: Could not read interface wlan1 flags: No such device
Dec 23 00:57:55 raspberrypi hostapd[2402]: nl80211 driver initialization failed.
Dec 23 00:57:55 raspberrypi hostapd[2402]: wlan1: interface state UNINITIALIZED->DISABLED
Dec 23 00:57:55 raspberrypi hostapd[2402]: wlan1: AP-DISABLED
Dec 23 00:57:55 raspberrypi hostapd[2402]: wlan1: CTRL-EVENT-TERMINATING
Dec 23 00:57:55 raspberrypi hostapd[2402]: hostapd_free_hapd_data: Interface wlan1 wasn't started
Dec 23 00:57:55 raspberrypi systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
 Subject: Unit process exited
 Defined-By: systemd
 Support: https://www.debian.org/support

An ExecStart= process belonging to unit hostapd.service has exited.

The process' exit code is 'exited' and its exit status is 1.
Dec 23 00:57:55 raspberrypi systemd[1]: hostapd.service: Failed with result 'exit-code'.
 Subject: Unit failed
 Defined-By: systemd
 Support: https://www.debian.org/support
 The unit hostapd.service has entered the 'failed' state with result 'exit-code'.
Dec 23 00:57:55 raspberrypi systemd[1]: Failed to start Access point and authentication server for Wi-Fi and Ethernet.
Subject: A start job for unit hostapd.service has failed
Defined-By: systemd

pi@raspberrypi:~ $ ifconfig -a
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.2.1  netmask 255.255.255.0  broadcast 192.168.2.255
        ether <removed>  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500 (currently connected)
        inet 192.168.1.232  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 <removed>  prefixlen 64  scopeid 0x0<global>
        inet6 <removed> prefixlen 64  scopeid 0x20<link>
        inet6 <removed>  prefixlen 128  scopeid 0x0<global>
        ether <removed> txqueuelen 1000  (Ethernet)
        RX packets 804  bytes 104407 (101.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 723  bytes 118527 (115.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 25  bytes 3878 (3.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 25  bytes 3878 (3.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.231  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 <removed>  prefixlen 64  scopeid 0x0<global>
        inet6 <removed>  prefixlen 64  scopeid 0x20<link>
        inet6 <removed>  prefixlen 128  scopeid 0x0<global>
        ether <removed> txqueuelen 1000  (Ethernet)
        RX packets 514  bytes 86181 (84.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 61  bytes 8671 (8.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

pi@raspberrypi:~ $

Someone experienced with scripts just needs to figure out why in setup.sh the  "Write XLink Kai launch script" doesn't get pass Launching XLink Kai

osaeed18, 2022-12-23:
ok after some testing here's what I found

I uncommented Write XLink Kai launch script from the setup.sh

```
Write XLink Kai launch script
cat <<'EOF' > /home/pi/launchkai.sh
echo "Checking for XLink Kai updates"
sudo apt-get install xlinkkai -y
echo "Launching XLink Kai"
while true; do
screen -dmS kai kaiengine
sleep 5
done
EOF

chmod 755 /home/pi/launchkai.sh
```
In setup.sh, I replaced the line that says /home/pi/launchkai.sh with sudo kaiengine
I reran the setup.sh and it successfully rebooted the pi.
We can see the changes to the line "#Start services" in https://github.com/toolboc/psx-pi-smbshare/commit/4cac72d84d95e405269d45b6aaeac12de8579c10

```
# Start services
sudo /usr/local/bin/samba-init.sh
sudo /home/pi/wifi-to-eth-route.sh
sudo /home/pi/setup-wifi-access-point.sh
ps3netsrv++ -d /share/
sudo kaiengine
```
Hope that helps

SpikeTrike, 2023-02-16:
> ok after some testing here's what I found
> 
> I uncommented Write XLink Kai launch script from the setup.sh
> 
> ```
> Write XLink Kai launch script
> cat <<'EOF' > /home/pi/launchkai.sh
> echo "Checking for XLink Kai updates"
> sudo apt-get install xlinkkai -y
> echo "Launching XLink Kai"
> while true; do
> screen -dmS kai kaiengine
> sleep 5
> done
> EOF
> 
> chmod 755 /home/pi/launchkai.sh
> ```
> 
> In setup.sh, I replaced the line that says /home/pi/launchkai.sh with sudo kaiengine I reran the setup.sh and it successfully rebooted the pi. We can see the changes to the line "#Start services" in [4cac72d](https://github.com/toolboc/psx-pi-smbshare/commit/4cac72d84d95e405269d45b6aaeac12de8579c10)
> 
> ```
> # Start services
> sudo /usr/local/bin/samba-init.sh
> sudo /home/pi/wifi-to-eth-route.sh
> sudo /home/pi/setup-wifi-access-point.sh
> ps3netsrv++ -d /share/
> sudo kaiengine
> ```
> 
> Hope that helps

Hello, I am trying to get this to work on my Pi 4 (4GB model), and when attempting to use this solution of commenting out the Xlink Kai launch script, It gets stuck after:

interface=wlan1
bind-dynamic
dhcp-range=10.254.0.2,10.254.100.12h
Failed to start hostapd.service: Unit hostapd.service is masked.
XLink Kai v7.4.43 build 606669993 Linux ARMHF (c) 2022 Team XLink https://www.teamxlink.co.uk/

Config file: ./kaiengine.conf
Web interface is ready, connect to http://127.0.0.1:34522/ or http://127.0.1.1:34522/
Downloading Language file en.xml
Initializing XLink Kai engine...

Downloading Configuration OK
Updating webui
Kai Engine: The engine has started
webui downloaded 
webui verified OK

 Did you ever run into this issue and/or get a version of this script running without [XLink Kai completely](https://github.com/toolboc/psx-pi-smbshare/issues/60)? I also am not planning on playing online so I do not even need the service. Thanks. :)

PR #62 (2023-02-21): setup.sh: add kailauncher to the background
Moving the launch script of XLink Kai to the background should fix #50 #60 and #41 from associated installation errors in my testing.

Edit: The release of Kaiengine 7.4.44 should address #41 in our testing.
closed 2023-02-21 by toolboc