toolboc/psx-pi-smbshare #57: Pi 4 memory size?

cystbear, 2022-10-27:
Hello,

How many gigs of memory is best for **Raspberry Pi 4 Model B** to use with _psx-pi-smbshare_
There are next options: 1GB, 2GB, 4GB, 8GB.

What I have to choose?

osaeed18, 2022-12-18:
I have a Raspberry Pi 4 B 2GB model.
psx-pi-smbshare isn't that memory intensive. I have a network folder attached and at most it uses 100+MB of memory.
You should be fine with 1GB, but if you use custom scripts you might run out of physical memory. 
Raspberry Pi 4 B 2GB model seems to be the sweet spot. 
As you can see below, there is still quite a lot of available memory left.

pi@ps2pi:~ $ free -m
               total        used        free      shared  buff/cache   available
Mem:            1872          71        1666           1         134        1718
Swap:             99           0          99
pi@ps2pi:~ $

cystbear, 2022-12-19:
@osaeed18 thanks a lot!

closed 2022-12-19 by cystbear