toolboc/psx-pi-smbshare #82: Pi Zero 2W compatibility

samanvayu, 2023-12-03:
I had researched about psx-pi-smbshare and it seems to be compatible with pi zero. However when I tried running it with pi zero 2 w, it appears to be stuck at the loading screen. The same image runs properly for a pi 3. Am I missing something and if not, will this issue be fixed in the future?

Stefkeys, 2023-12-31:
Bumping this. Experiencing the same issue with Sudo on a Pi4

samanvayu, 2023-12-31:
I can somewhat help you with it. I researched more and turns out its a pi update thing because of which they had to change the script. Unfortunately there's conflicting stuff so it doesn't work. You can look up earlier versions of this script and try running that by simply copying and making your own .sh files. However, I faced an issue where the stuff that the .sh files were no longer downloading, maybe the servers were down I don't know and you can try that. You can always look up the BDM network thing it's just that you have to set up a server everytime you try playing and not something that you can just put together and forget about it. I can't do BDM because mine is a Pi Zero. However, the best option that I found was a bitbuilt link which you can use to set up a proper SMBshare yourself without all this pizzazz. It works well and what I'm presently using. Only caveat is I don't know of it'll work for PS3 and stuff. Here's the link for the guide: https://bitbuilt.net/forums/index.php?threads/ps2-smb-with-a-raspberry-pi-or-how-to-load-games-at-near-full-speed.1780/. BIG UPDATE that's not there in the bitbuilt link. You have to update something in the global setting in smb.conf : client min protocol = NT1
server min protocol = NT1. Also, this is something I didn't look at but you can always look at retroNAS. That's it. Lemme know if you need any other guidance.