toolboc/psx-pi-smbshare #83: Cannot open smb share problem, ps2-rapberry pi 4

vivek57-github, 2023-12-11:
i bought a rapsberry pi 4 solely for to use it for connecting to ps2, but after setting all the network settings on the ps2 correctly matching the given instructions, the opl v1.1.0 has been saying "cannot open smb share". i was struck here and please some one help on this one !

ivanEsteiro, 2023-12-19:
Hi
I'm dealing with some similar issues with pi 4 and opl1.0.0. So far im figuring that the Pi is ok with some storage devices but not with others. 
A 64gb usb in exfat format is being taken withouth issue for now. 
My 124gb ssd connected from Sata to usb is sometimes giving me that 303 error while some other times doesnt. Tried exfat, fat32, and nts the result is the same.
I'll be tweaking around a few more options and see how they react as well as checking some past threads. Will update if I find a breakthrough

georgewoodall82, 2024-01-22:
I made a fork which fixes most usb issues https://github.com/georgewoodall82/psx-pi-smbshare-updated

but you need to flash raspberry pi os to your pi, and then run the install script instead of using a pre-made image.

It switches from pmount to udisks2, because pmount is what was previously used to mount the usb which hasn't been updated in 4 years, and has many problems (for me i could only mount ntfs drives). And for me, the speed went up ALOT.

I also made a pull request but i don't think it will get merged anytime soon.

StefiStarlite, 2024-02-01:
Even using the fork I still have 303 issues (constantly, not intermittent. Is there a way to verify the USB was mounted correctly for the share to work on the Pi OS desktop?

georgewoodall82, 2024-02-01:
I recommend something big like 128 or 64 KiB for large files for speed.
Also, did you use the setup script on the fork? because currently the downloads are just the old versions so you need to use the setup script.

StefiStarlite, 2024-02-01:
I installed using the script from your project fork, I triple checked before running the script to make sure your fork was the source and not the previous versions.

Skuzee, 2025-05-28:
> I made a fork which fixes most usb issues https://github.com/georgewoodall82/psx-pi-smbshare-updated
>
> but you need to flash raspberry pi os to your pi, and then run the install script instead of [...]

Is it possible for me to update to using this version on a pi4 that already has the original script installed? My issue is that I can connect to the SD card share folder with the ps2, but when I plug in an external HDD, it says that the smb share cannot connect, and I can't see any games. Not sure if this is my issue, or if I have something wrong with my drive setup.

georgewoodall82, 2025-05-28:
> Is it possible for me to update to using this version on a pi4 that already has the original script installed? My issue is that I can connect to the SD card share folder with the ps2, but when I [...]

Sure, try deleting old files then running the script on my repo:
```
sudo systemctl stop smbd
sudo rm -rf /share
sudo rm /usr/local/bin/samba-init.sh
rm /home/pi/samba-init.sh
rm /home/pi/wifi-to-eth-route.sh
rm /home/pi/setup-wifi-access-point.sh
rm /home/pi/automount-usb.sh
rm /home/pi/launchkai.sh
crontab -r
```