toolboc/psx-pi-smbshare #85: Installation on base Ubuntu/Debian

megacorvega, 2024-01-07:
The following changes will allow installation for base Ubuntu/Debian installs

1. run `sudo apt install g++ net-tools`
2. For Debian and Ubuntu remove `exfat-utils` from the deps install section of the `automount-usb.sh` script
3. Check ethernet and wlan interface names with `ifconfig` then change the `eth` and `wlan` entries in `wifi-to-eth-route.sh`
4. In the `setup.sh` script I commented out the crontab line, and manually added the startup scripts to the root cron. Using `sudo` in the user's crontab asks for a password on boot by default, and will not work on reboot.

`sudo crontab -e` 
```
# root crontab
# 1. log outputs aren't necessary but can be helpful
# 2. I also don't use the wifi access point or xlinkkai in my install, but those can be added here if desired

@reboot bash /usr/local/bin/samba-init.sh > /home/pi/samba-init.log 2>&1
@reboot bash /home/pi/wifi-to-eth-route.sh > /home/pi/wifi-to-eth-route.log 2>&1
```
These changes *should* allow you to run the base `setup.sh` script, but I still recommend manually running each line to see if something doesn't get setup correctly, and checking each of the services start scripts as well.

Not sure if these modifications want to get added to the repo/wiki, but they're here if anyone comes looking.

Trevo525, 2024-01-08:
You should consider forking this and making your changes. Maybe toolboc would accept a pull request even?

megacorvega, 2024-02-29:
@Trevo525 Yeah that's a good idea. I'll get that going.