toolboc/psx-pi-smbshare #92: Error Code 300 PLEASE DO NOT IGNORE

SelectWall79786, 2024-05-06:
Hello, I have set this up before on a Pi 4, but I took it apart, and formatted my SD card to do a different project. I have psx-pi-smbshare installed. I have my PS2 hooked up with a Ethernet wire into my Pi. I did copy the settings provided within the screenshot for the OPL network settings. Right now my Pi keeps requesting a Ethernet address from my PS2, but it fails to address one. I am lost. I have no idea what to do to fix, and solve my issue. I want to say last time I changed the numbers for the network settings, but I have no idea what, and why I decided to change them, to make it work. I really think this is a fantastic project, I just want to join in on the fun within this project for my PS2. Thank you, and I appreciate the time you spend helping me figure this out.

ja50nR0mAn, 2024-05-07:
Hi, also having huge challenges getting this to work on Pi4. 

Try this tutorial: 
https://tuliocalil.com/ps2-smb-raspberry-pi4-in-2024/

It worked for me but not as well as the pre-installed Pi3B image on here. 

Let me know if you are able to get it to work on the Pi4 seamlessly.

georgewoodall82, 2024-06-08:
@SelectWall79786 Try renaming your usb to nothing, just a blank name.
If that doesn't work, try the latest version of OPL from here: https://github.com/ps2homebrew/Open-PS2-Loader/releases

mraptakis, 2024-09-06:
Hello everyone!

I have followed the tutorial from [https://tuliocalil.com/ps2-smb-raspberry-pi4-in-2024/](https://tuliocalil.com/ps2-smb-raspberry-pi4-in-2024/) and adjust appropriately with my setup. I'm using a USB stick for the games that I have already tested by connecting it directly to PS2.

But, I am getting error 300.

Any ideas?

Thank you in advance.

georgewoodall82, 2024-09-06:
Try disconnecting the USB from the raspberry pi and then connecting. if it connects successfully, without errors and without showing any games, you can try using my updated version with the instructions on https://github.com/georgewoodall82/psx-pi-smbshare-updated

if you still get the error, make sure your using the latest OPL build from here https://github.com/ps2homebrew/Open-PS2-Loader/releases
old versions of OPL have caused this same issue multiple times before.

mraptakis, 2024-09-09:
During troubleshooting, I realize something that, maybe, is important! I have installed FreeMCBoot in a Memory Card but I have used the FunTune program and I cannot update the OPL. I believe that this is the reason that I can't make it work.