toolboc/psx-pi-smbshare #98: Error Code 300 Potential Solution?

co505, 2024-09-15:
Hi,

I was struggling to get my Raspberry Pi 4B to communicate with the PS2 via SMB last night, kept getting thrown an "Error 300: Cannot connect to SMB server".

I followed the advice to install the latest release of OPL and the updated guide, but still no success. Everything seemed configured as it should. 

I saw the `ifconfig` output did not list an `inet` address for 'eth0'. If I set this manually, using 'sudo ifconfig eth0 192.168.2.1' on the Pi then my issues were resolved and the PS2 was able to use SMB successfully. I was able to access my game list on the recent and older versions of OPL using this method, so the OPL version did not seem to matter. 

I haven't looked into why fix worked, but I thought it would be a quick solution for people to test if they were struggling with Error 300 and the 'inet' address was missing for 'eth0'.

Happy to dive into it more if it would be of some use, and apologies if you were already aware of this.

Thanks.

egorkaru, 2024-09-16:
Are you using usb drive?

If yes, try to update `Path` parameter under `[share]` section in `/usr/local/bin/automount.sh` from:
```sh
Path = /media/userplaceholder/$UUID
```

to

```sh
Path = /media/userplaceholder/$FS_LABEL
```

Dadibom, 2024-09-22:
i need to manually run wifi-to-eth-route.sh after every boot or it won't work.

chornbec, 2024-10-15:
> i need to manually run wifi-to-eth-route.sh after every boot or it won't work.
This worked for me. I was using a Raspberry Pi 4 and followed [the steps](https://github.com/toolboc/psx-pi-smbshare/issues/10) to install Raspbian Lite first.